openembedded/meta-openembedded #864: dm-verity checking fails while using sparse format image.

bajerwitharm, 2024-08-22:
Seems dm-verity checking fails while using sparse format image. I think the problem is a commit:

https://git.openembedded.org/meta-openembedded/commit/meta-oe/classes/image_types_sparse.bbclass?h=master-next&id=9862a017fa7f88424f0670ba89af58e5051550b0

which in my opinion changes the content of ext4 image included in the wic file. Without `-s` option in `img2simg` call the image works wihtout any errors.

As I see without `-s` image size is much bigger, but still smaller than without `sparse` formation. 

Is there a way to keep `sparse` format and use `dm-verity`?

quaresmajose, 2024-08-23:
you can propose a patch to make the `-s` argument configurable and disable the arg for `dm-verity`, something like:

```patch
SPARSE_BLOCK_SIZE ??= "4096"
+SPARSE_IMG2SIMG_ARGS ??= "-s"

CONVERSIONTYPES += "sparse"
 CONVERSION_CMD:sparse() {
     INPUT="${IMAGE_NAME}${IMAGE_NAME_SUFFIX}.${type}"
     truncate --no-create --size=%${SPARSE_BLOCK_SIZE} "$INPUT"
-    img2simg -s "$INPUT" "$INPUT.sparse" ${SPARSE_BLOCK_SIZE}
+    img2simg ${SPARSE_IMG2SIMG_ARGS} "$INPUT" "$INPUT.sparse" ${SPARSE_BLOCK_SIZE}
 }
 CONVERSION_DEPENDS_sparse = "android-tools-native"
```

bajerwitharm, 2024-08-23:
Sounds reasonable for me. For sure it will help with my problem.