openembedded/meta-openembedded #868: Ostree

formiano, 2024-09-10:
ERROR: Unable to parse build-enviroment/meta-openembedded/meta-oe/recipes-extended/ostree/ostree_2024.7.bb
Traceback (most recent call last):
  File "build-enviroment/bitbake/lib/bb/pysh/pyshlex.py", line 600, in PLYLexer._parse(eof=True):
                 elif self._state==self.ST_QUOTED:
    >                self._parse_quoted(eof)
                 elif self._state==self.ST_HEREDOC:
  File "/build-enviroment/bitbake/lib/bb/pysh/pyshlex.py", line 688, in PLYLexer._parse_quoted(eof=True):
                 
    >        wtree, remaining = self._wordlexer.add(input, eof)
             self._wordlexer = None
  File "/build-enviroment/bitbake/lib/bb/pysh/pyshlex.py", line 146, in WordLexer.add(data=

shr-project, 2024-09-10:
Parses fine here, isn't it something caused by oe-alliance changes?

formiano, 2024-09-10:
RROR: build-enviroment/meta-openembedded/meta-oe/recipes-extended/ostree/ostree_2024.7.bb: Error during parse shell code, the last 5 lines are:TA:  0:01:26
		fi
	else
		bbfatal "no configure script found at $cfgscript"