openembedded/meta-openembedded #871: meta-oe: gpsd depends on meta-python unconditionally

lumag, 2024-09-12:
Adding just meta-oe results in the following error from bitbake:

```
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'python3-pyserial' (but ../meta-openembedded/meta-oe/recipes-navigation/gpsd/gpsd_3.25.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'python3-pyserial' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python3-pyserial']
NOTE: Runtime target 'gpsd' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gpsd', 'python3-pyserial']
...
```

quaresmajose, 2024-09-12:
maybe we can make it conditional with a PACKAGECONFIG, disabled by default

```
PACKAGECONFIG[ubxtool] = ",,python3-pyserial"
```

lumag, 2024-09-13:
Well, there is also a second issue: if there is a CI for meta-oe, it needs to be updated to catch such issues. I unfortunately don't know enough details of the CI setup to approach this problem.

quaresmajose, 2024-09-23:
I send a fix for this: https://lists.openembedded.org/g/openembedded-devel/message/112456

quaresmajose, 2024-09-23:
> Well, there is also a second issue: if there is a CI for meta-oe, it needs to be updated to catch such issues. I unfortunately don't know enough details of the CI setup to approach this problem.

I believe the yocto-check-layer can catch this type of things.

quaresmajose, 2024-09-27:
fixed https://github.com/openembedded/meta-openembedded/commit/eb731aa8660b280a15684dc81954dbcd4d54fdb6

lumag, 2024-09-27:
It is not. With just `meta-oe` added and with the default configuration I'm still getting:

```
ERROR: Nothing RPROVIDES 'python3-pyserial' (but /home/lumag/Projects/RPB/build-rpb/conf/../../layers/meta-openembedded/meta-oe/recipes-navigation/gpsd/gpsd_3.25.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'python3-pyserial' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python3-pyserial']
ERROR: Required build target 'gpsd' has no buildable providers.
Missing or unbuildable dependency chain was: ['gpsd', 'python3-pyserial']
```

quaresmajose, 2024-09-27:
well! in reality I didn't remove the problem.
```patch
 RDEPENDS:python3-pygps = " \
   python3-core \
   python3-io \
-  python3-pyserial \
   python3-threading \
   python3-terminal \
   gpsd \
   python3-json"
```
Thanks for notifying me
I will have to come back to this again

lumag, 2024-09-27:
Please also check that RDEPENDS get addded back if pyserial is enabled

quaresmajose, 2024-09-30:
v2 merged https://github.com/openembedded/meta-openembedded/commit/09563d1b395b36d5e66a13b0ac67f1bdf2a761eb

lumag, 2024-10-06:
Seems to be working now, thank you!

closed 2024-10-06 by lumag