openembedded/meta-openembedded #874: opensc: multiple security vulnerablities require backports?

citypw, 2024-09-25:
There are multiple security vulnerabilities are fixed in OpenSC v0.26-rc1:

https://github.com/OpenSC/OpenSC/releases/tag/0.26.0-rc1

kraj, 2024-09-26:
@citypw I see that its still in RC stage. Once 0.26 final is released, we need to upgrade the recipe.

citypw, 2024-09-26:
@kraj do you have plan to backport it to other branches like Kirkstone? There are some security backports still missing in Kirkstone:
https://github.com/OpenSC/OpenSC/wiki/OpenSC-security-advisories#opensc-security-advisories

kraj, 2024-09-26:
> @kraj do you have plan to backport it to other branches like Kirkstone? There are some security backports still missing in Kirkstone: https://github.com/OpenSC/OpenSC/wiki/OpenSC-security-advisories#opensc-security-advisories

Usual policy is no major version upgrades into release branches. It will surely be in master when it happens.

citypw, 2024-09-27:
I understand the point. I saw some branches like Kirkstone did the security backports for OpenSC previously:
https://github.com/openembedded/meta-openembedded/blob/kirkstone/meta-oe/recipes-support/opensc/opensc_0.22.0.bb#L17C1-L25C43

It's still missing a couple of known vulnerabilities with CVE numbers. I'm curious what's the backport criteria. Will all CVEs backport to the branches or just some CVEs with higher impact?

kraj, 2024-09-27:
it really depends upon contributors.

citypw, 2024-09-28:
Okidoki, a PR with two backports: https://github.com/openembedded/meta-openembedded/pull/876

kraj, 2026-10-26:
PR is merged.

closed 2026-10-26 by kraj