openembedded/meta-openembedded #875: scarthgap: samba 4.19.8 meets configure break as libldb version is not found

junzhuimx, 2024-09-26:
As samba is upgrade to 4.19.8, the depended libldb should also upgrade to v2.8.1.
Recently, do_configure failed as:
| Checking for system pyldb-util (>=2.8.1 <=2.8.999)                                : not found
| ERROR: System library pyldb-util of version 2.8.1 not found, and bundling disabled

junzhuimx, 2024-10-05:
@kraj The issue has caused "samba" build break for 1 week on branch "scarthgap", and the commit 09f8ef224 should cherry-picked from branch master

kraj, 2024-10-05:
@junzhuimx thanks for reporting, I think it needs to come to stable maintainer @akuster 's notice. It would be better if you can send a patch which is backported and tested on scarthgap. That will make his life easier too.

akuster, 2024-10-06:
I just cherry-picked it. it should land shortly

junzhuimx, 2024-10-08:
@kraj  and @akuster 
the PR https://github.com/openembedded/meta-openembedded/pull/879 can be merged to fix this issue.

kraj, 2026-10-26:
fixed with 5d3fb27a4edff7ef13280a5ab66e638fd4df7519 on scarthgap

closed 2026-10-26 by kraj