openembedded/meta-openembedded #895: Node.js version 22.11.0 build failure on ubuntu 20 with gcc 9

sangmo-kang, 2024-11-11:
Dear, Maintainer.

After upgrading nodejs to 22.11.0(https://github.com/openembedded/meta-openembedded/commit/6652b031d7d4d9cabf16d481f3b82115f7d8e15a), `bitbake nodejs-native` fails with the following error:
g++: error: unrecognized command line option ‘-std=gnu++20’; did you mean ‘-std=gnu++2a’?

System infomation:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal
$ gcc --version
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Log Details:
```
|   g++  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/simdutf/deps/simdutf/simdutf.o ../deps/simdutf/simdutf.cpp '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/simdutf  -pthread -Wall -Wextra -Wno-unused-parameter -m64 -O3 -fno-omit-frame-pointer -fno-rtti -fno-exceptions -fno-strict-aliasing -std=gnu++20 -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/simdutf/deps/simdutf/simdutf.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
|   gcc  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/threadpool.o ../deps/uv/src/threadpool.c '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_GNU_SOURCE' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/uv/include -I../deps/uv/src  -pthread -Wall -Wextra -Wno-unused-parameter -fvisibility=hidden -g --std=gnu89 -Wall -Wextra -Wno-unused-parameter -Wstrict-prototypes -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/threadpool.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
|   gcc  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/timer.o ../deps/uv/src/timer.c '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_GNU_SOURCE' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/uv/include -I../deps/uv/src  -pthread -Wall -Wextra -Wno-unused-parameter -fvisibility=hidden -g --std=gnu89 -Wall -Wextra -Wno-unused-parameter -Wstrict-prototypes -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/timer.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
|   gcc  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/uv-data-getter-setters.o ../deps/uv/src/uv-data-getter-setters.c '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_GNU_SOURCE' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/uv/include -I../deps/uv/src  -pthread -Wall -Wextra -Wno-unused-parameter -fvisibility=hidden -g --std=gnu89 -Wall -Wextra -Wno-unused-parameter -Wstrict-prototypes -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/uv-data-getter-setters.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
|   gcc  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/uv-common.o ../deps/uv/src/uv-common.c '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_GNU_SOURCE' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/uv/include -I../deps/uv/src  -pthread -Wall -Wextra -Wno-unused-parameter -fvisibility=hidden -g --std=gnu89 -Wall -Wextra -Wno-unused-parameter -Wstrict-prototypes -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/uv-common.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
|   gcc  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/version.o ../deps/uv/src/version.c '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_GNU_SOURCE' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/uv/include -I../deps/uv/src  -pthread -Wall -Wextra -Wno-unused-parameter -fvisibility=hidden -g --std=gnu89 -Wall -Wextra -Wno-unused-parameter -Wstrict-prototypes -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/version.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
|   gcc  -o /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/unix/async.o ../deps/uv/src/unix/async.c '-D_GLIBCXX_USE_CXX11_ABI=1' '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-D__STDC_FORMAT_MACROS' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D_GNU_SOURCE' -I/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -I../deps/uv/include -I../deps/uv/src  -pthread -Wall -Wextra -Wno-unused-parameter -fvisibility=hidden -g --std=gnu89 -Wall -Wextra -Wno-unused-parameter -Wstrict-prototypes -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer  -MMD -MF /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/.deps//home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/node-v22.11.0/out/Release/obj.host/libuv/deps/uv/src/unix/async.o.d.raw  -isystem/home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/recipe-sysroot-native/usr/include -O2 -pipe -c
[... cut ...]
ERROR: Task (virtual:native:/home/sangmo.kang/testoe/repo/meta-openembedded/meta-oe/recipes-devtools/nodejs/nodejs_22.11.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 488 tasks of which 462 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/sangmo.kang/testoe/repo/meta-openembedded/meta-oe/recipes-devtools/nodejs/nodejs_22.11.0.bb:do_compile
    log: /home/sangmo.kang/testoe/build/tmp/work/x86_64-linux/nodejs-native/22.11.0/temp/log.do_compile.10210
Summary: There were 2 ERROR messages, returning a non-zero exit code.

```

shr-project, 2024-11-11:
The default gcc in ubuntu 22.04 is 11.2.0, try that instead of 9.4.0 you're using for some reason.
```
docker-lge @ ~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04 LTS
Release:        22.04
Codename:       jammy
docker-lge @ ~ $ gcc --version
gcc (Ubuntu 11.2.0-19ubuntu1) 11.2.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

sangmo-kang, 2024-11-11:
Thanks for your quick response!
I found that `std=gnu++20` was not working in gcc 9 and below.
However, it seems that Yocto's gcc requirements are 8.0 or greater(https://docs.yoctoproject.org/dev/ref-manual/system-requirements.html#required-git-tar-python-make-and-gcc-versions). 
I'm curious if there is a way to solve this problem while maintaining gcc version 9.

shr-project, 2024-11-11:
Why are you using gcc-9? isn't it configured, just because you were using 18.04 (which had gcc-7 by default) before and you forgot to remove it when upgrading to 22.04?

sangmo-kang, 2024-11-11:
I have set up a own development environment with Ubuntu 20 and GCC 9. I was just curious about the support for GCC 9. I will consider upgrading my development environment to Ubuntu 22.

closed 2024-11-11 by sangmo-kang