openembedded/meta-openembedded #919: [scarthgap] libcamera_0.2.0.bb is broken

thomas-roos, 2025-01-15:
libcamera_0.2.0.bb is broken on scarthgap, backporting (drop in replacement) of libcamera_0.4.0.bb fixed this for me.
Related, same issues: https://github.com/raspberrypi/rpicam-apps/issues/654
Was testing on a Raspberry Pi 2 Zero W with Cam module V3.
I suggest to backport libcamera_0.4.0.bb from master.
This worked seamless for me without any additional change.
Thank you!

kraj, 2025-01-15:
@thomas-roos thats perhaps fine to have an exception, would you mind creating a PR or send the patch to ML ?

thomas-roos, 2025-01-16:
@kraj will do. Ty!

thomas-roos, 2025-01-20:
Did send it: https://lists.openembedded.org/g/openembedded-devel/message/114996
Think I did something wrong with the command?
`git send-email --to openembedded-devel@lists.openembedded.org --subject-prefix='meta-multimedia] [scarthgap] [PATCH' 0001-libcamera-backport-0.4.0-from-master-next.patch
`
-> what does produce the correct header?

thomas-roos, 2025-01-21:
Ok, learned that I need to create the patch with correct subject line and correct escaping.
This should do it:
```
git format-patch -1 --subject-prefix='meta-multimedia] [scarthgap] [PATCH'
git send-email --to openembedded-devel@lists.openembedded.org 0001-libcamera-backport-0.4.0-from-master-next.patch
```

Did send again: https://lists.openembedded.org/g/openembedded-devel/message/115002
Thank you!

kraj, 2026-10-26:
fixed with 0bd3f27be016b7f0935b2ed5b0dc481689cf10ef on scarthgap

closed 2026-10-26 by kraj