openembedded/meta-openembedded #932: [libspdm] unsupported architecture 'mipsel'

fairbird, 2025-02-07:
ERROR: /home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/meta-openembedded/meta-oe/recipes-support/libspdm/libspdm_3.6.0.bb: unsupported architecture 'mipsel'
ERROR: /home/raed/build_image/My-Openpli/openpli-dreambox-oe-core/meta-openembedded/meta-oe/recipes-support/libspdm/libspdm_3.6.0.bb: unsupported architecture 'mipsel'

OldManYellsAtCloud, 2025-02-07:
Yeah, it's like that unfortunately. The built project itself doesn't support this arch, so it is excluded from the recipe also: https://github.com/DMTF/libspdm/blob/main/CMakeLists.txt#L50

fairbird, 2025-02-07:
From my side I have solved with to add These two lines in (mips32el.inc) files

> BBMASK += "/meta-openembedded/meta-oe/recipes-support/libspdm/"
BBMASK += "/meta-openembedded/meta-oe/dynamic-layers/clang-layer/recipes-devtools/spdm-utils/"

kraj, 2025-02-08:
you can add

```
COMPATIBLE_HOST:mipsarcho32 = "null"
```

to the recipe `meta-oe/recipes-support/libspdm/libspdm_3.6.0.bb`
and also to `meta-oe/dynamic-layers/clang-layer/recipes-devtools/spdm-utils/spdm-utils_0.7.2.bb`

I will accept such a patch.

kraj, 2026-10-25:
should be fixed with f0f690a64d on master

closed 2026-10-25 by kraj