openembedded/meta-openembedded #966: lmsensors build broken on scarthgap if sensord is disabled

kone-tlammi, 2025-05-20:
`lmsensors_3.6.0.bb` do_install fails with
```
make: *** No rule to make target 'rrd.h', needed by 'prog/sensord/rrd.rd'.  Stop.
```
when `PACKAGECONFIG = ""`

The breaking change seems to be the current branch head (e92d0173a80ea7592c866618ef5293203c50544c).
The issue does not happen in the previous commit (01eb9fb907892da37d3d88c37d1191151e2a9e31) or
if `PACKAGECONFIG = "sensord"`

OldManYellsAtCloud, 2025-05-20:
There is a [fix](https://lore.kernel.org/openembedded-devel/20250505095620.14907-1-l.anderweit@phytec.de/) floating around for this, but it hasn't found its way in the repo yet. But I think it should happen in a few days, upon the next Scarthgap merge

kraj, 2026-10-26:
fixed on scarthgap with 154b5010be669527f850b3b3f3162ccd97f639f9

closed 2026-10-26 by kraj