openembedded/meta-openembedded #989: [Scarthgap] packagegroup-meta-oe-support fails when x11 is not enabled due to remmina

JosephMillsAtWork, 2025-08-25:
## Issue:

packagegroup-meta-oe-support fails when x11 is not enabled due to remmina needs of x11. I am sure that there are more. Im gonna go through them and comment as needed.

From meta-openembedded/meta-oe/recipes-support/remmina/remmina_1.4.34.bb

```bb
REQUIRED_DISTRO_FEATURES = "x11"
```

Suggested fix

```patch
diff --git a/meta-oe/recipes-core/packagegroups/packagegroup-meta-oe.bb b/meta-oe/recipes-core/packagegroups/packagegroup-meta-oe.bb
index 71f74287c7..dd352cfcdc 100644
--- a/meta-oe/recipes-core/packagegroups/packagegroup-meta-oe.bb
+++ b/meta-oe/recipes-core/packagegroups/packagegroup-meta-oe.bb
@@ -826,7 +826,7 @@ RDEPENDS:packagegroup-meta-oe-support ="\
     monit \
     mscgen \
     libsmi \
-    remmina \
+    ${@bb.utils.contains("DISTRO_FEATURES", "x11", "remmina", "", d)} \
     neon \
     nmon \
     libjs-jquery-icheck \

```

JosephMillsAtWork, 2025-08-25:
same with 

packagegroup-meta-oe  -> ```pcp_6.0.5.bb``` 

packagegroup-meta-python  -> ```python3-pyatspi_2.46.1.bb``` 

```patch
@@ -301,7 +301,7 @@ RDEPENDS:packagegroup-meta-python3 = "\
     python3-py-ubjson \
     python3-pyalsaaudio \
     python3-pyasn1-modules \
-    python3-pyatspi \
+    ${@bb.utils.contains_any("DISTRO_FEATURES", "x11 directfb", "python3-pyatspi", "", d)} \
     python3-pyaudio \
     python3-pybind11 \
     python3-pybind11-json \
```

kraj, 2026-10-26:
@JosephMillsAtWork its fixed on master with 268b24cf14f49e93edb9afbdf3fc6feb799bc0c8 please send a backport request for scarthgap to mailing list and close it.