openembedded/meta-openembedded #994: [scarthgap] Poppler las commit breaks build

koansoftware, 2025-09-04:
After commit [c8a1b90](https://github.com/openembedded/meta-openembedded/commit/c8a1b909ec78973c9de668d92d9b39a5b005c6d8)
The following two patches are breaking the build.

```
  file://CVE-2025-52886-0001.patch \
  file://CVE-2025-52886-0002.patch \
```

For example
```
poppler/23.04.0/poppler-23.04.0/qt5/src/poppler-annotation.cc:5082:2: error: expected '}' at end of input
|  5082 | }
```

OldManYellsAtCloud, 2025-09-04:
Do you use some non-default config? I have just compiled it successfully with default config.

OldManYellsAtCloud, 2025-09-04:
Yes, that patch seems to have some typos... I have submitted a patch as an attempt to fix it: https://patchwork.yoctoproject.org/project/oe/patch/20250904150540.3041738-1-skandigraun@gmail.com/
If you have the chance, feel free to give it a try. (Not sure however when exactly will it show up in the repo)